badgeteam/mch2022-badge-hardware #18: RGB LED pinout doesn't match either iCEBreaker or Upduino

xesxen, 2020-12-06:
We match with neither of these boards. To make things easier/more consistent for users I suggest we stick to either of these two (I would favour the iCEBreaker personally since that's an opensource design).

Upduino uses ([PDF](https://raw.githubusercontent.com/gtjennings1/UPDuino_v2_0/master/PCB/UPDuino_v2_0_C_121217.pdf)):
```
RGB0: Red
RGB1: Green
RGB2: Blue
```
iCEBreaker on the other hand uses ([KiCAD](https://github.com/icebreaker-fpga/icebreaker/tree/master/hardware/v1.0e)):
```
RGB0: Blue
RGB1: Green
RGB2: Red
```
while Proto2 currently uses 
```
RGB0: Green
RGB1: Red
RGB2: Blue
```

sebastius, 2020-12-06:
That is something to fix in SW, right? Just a matter of pin definitions in some file? I haven't looked at it but pinwise, i'd go with what is easiest to route.

xesxen, 2020-12-06:
The pinconstraint is indeed just a plaintext file, but having it consistent with a popular board makes it easier for someone to just point to/use existing tutorials for RGB stuff :)

smunaut, 2020-12-06:
The pin constraint file isn't used for the RGB pins. Those have hardwired connections to the constant current driver Hard IP, you need to adapt the wiring in the verilog.

xesxen, 2020-12-06:
Are you sure? The iCEBreaker pcf clearly lists them as 'pins': https://github.com/osresearch/up5k/blob/master/icebreaker.pcf#L13-L16

smunaut, 2020-12-06:
Given I wrote the code in `nextpnr` to support the `SB_RGBA`  LED driver IP, yeah, I'm pretty sure.

They are in the PCF because you can still use those pins without using the constant current led drivers, as input or open drain (non current limited) outputs in which case the normal PCF applies. Also `nextpnr` will complain if they're not listed because ... reasons (pin locking is checked before we know what those pins are used for so at that point it doesn't know yet if they're used by the dedicated HW block)

sebastius, 2020-12-06:
> Given I wrote the code in `nextpnr` to support the `SB_RGBA` LED driver IP, yeah, I'm pretty sure.

Great to have you on board then! Thanks for helping out ☺️

xesxen, 2020-12-07:
Excuse me while I go eat my hat. Thanks for correcting me :)

renzenicolai, 2020-12-12:
Fixed, I chose to use the ICEbreaker pinout:

0: blue
1: green
2: red

closed 2020-12-12 by renzenicolai