badgeteam/mch2022-badge-hardware #20: RTC backup LDO

tachiniererin, 2020-12-06:
The AMS1117, U15 in the schematic won't be stable. It has a minimum load current requirement of 5mA typical, and worst case 10mA. In addition, it has a quiescent current of 5-11mA, which will drain the battery really fast.

In addition to that, the output capacitor is only 100nF, which is not enough to make it stable. In the datasheet it's specified with a 22µF tantalum (high ESR) capacitor, a ceramic capacitor has a magnitude less ESR in comparison.

[image]

sebastius, 2020-12-06:
Hmm yeah i agree on that the AMS1117 should be dropped for something better. Any suggestions?

tachiniererin, 2020-12-06:
Any requirements, e.g. does it have to be available from LCSC? I'll have a look in the morning.

sebastius, 2020-12-06:
We have a 'local agent' in China doing our purchasing so if LCSC doesnt have it, it might still be available. Our main goal is both cheap and efficient. Cheap in bulk (4000 badges are planned). LCSC would help/make things easier also for the prototype phase. It would mean that the LDO is readily available.

tachiniererin, 2020-12-07:
Depending on just how cost sensitive you are I would go either with TLV70433 (LCSC part number C94917) or as a cheaper alternative TLV74333 (LCSC part number C408972).

Both are stable with small MLCCs, so what they cost more you save in not needing an electrolytic (or worse, tantalum) capacitor.

sebastius, 2020-12-07:
it's all community money so the cheaper the better. I prefer not using tantalums, those are a fire hazard and contain a lot of slave labour.

tachiniererin, 2020-12-07:
A really low-cost alternative would be to supply the RTC through one or two silicon diodes to drop the battery voltage a bit. For example one or two 1N4148W in series, biased with a 200k resistor. Under no circumstances should Vbat then see more than 3.2V.
This then would have a static current consumption of 20µA including the RTC itself, which is even lower than the quiescent current of the cheaper LDO.

This solution would need to be tested though, because the characteristics of the diodes can vary a bit depending on the supplier. It might also be okay to reduce the bias current, if that's an optimization goal.

In total this would probably reduce the costs for this section to less than 1c per badge.

tachiniererin, 2020-12-07:
Adding a 3.3-V Zener would cancel out the temperature sensitivity of the previous solution:
[image]

This should be cheaper and better than any LDO for this particular job.

renzenicolai, 2020-12-12:
Added circuit with 3.3v zener instead of LDO

closed 2020-12-12 by renzenicolai