badgeteam/mch2022-badge-hardware #24: FPGA: Clock "domain" shared between LCD and I2S IOs due to pin choice

smunaut, 2020-12-06:
So ... another quirk of that FPGA, some pairs of IO share the clock line driving them.

Here `IOT_48b` and `IOT_49a` are such a pair, which is a bit annoying that one is assigned to logic driving the LCD and another to logic driving the audio since you might want to have those in different clock domains.

I would try to have `I2S_DATA` and `I2S_LR` in an IO site pair like 38 / 42  and then have `I2S_CLK` on its own on pin 28.  (yeah I know it's a bit further but that's the only pin in that bank that's not in an IO pair).

renzenicolai, 2020-12-12:
Would you be willing to create a pinout sheet that maps the signal names on the schematic to the pins to which they should be connected? The changes you mention sound very usefull but it's a bit much to figure out while you seem to know the ins-and-outs of it :smile:

smunaut, 2020-12-13:
Here's what I ended up with:

```
25  IOT_36b  LCD_FMARK
23  IOT_37a  LCD_WR
27  IOT_38b  LCD_D1
26  IOT_39a  LCD_D0
28  IOT_41a  I2S_CLK
31  IOT_42b  LCD_D2
32  IOT_43a  LCD_D3
34  IOT_44b  LCD_D4
37  IOT_45a  LCD_D5
35  IOT_46b  CLK
36  IOT_48b  I2S_DATA
43  IOT_49a  I2S_LR
38  IOT_50b  LCD_D6
42  IOT_51a  LCD_D7
```

Couple of comments:
* `I2S_DATA` / `I2S_LR` can be swapped to ease routing
* I ended up using the `36` / `43` pair for I2S since `36` is a corner pin, it'll be easier to place a via to escape it out of all the LCD signals around it and `43` is also more on the outside.
* `28` Should also be near a "void" created with the power connections on neighbor pins so a via to escape it outside the LCD signal to join up with the other I2S shouldn't be an issue.

PS: Not done with the PR, just working through the other issues

renzenicolai, 2020-12-13:
I've merged the PR for now, together with a change that didn't conflict.

Will you let me know once you're done with your work? I don't mind if it takes a bit longer, thank you for your help!

smunaut, 2020-12-13:
Oh, I'm done. The comment above was because I made that comment like ... 3 hours before submitting the PR.

renzenicolai, 2020-12-13:
Thank you for your work, this will make the badge a much more usefull toy :-)

I've sent you an invitation to the badge.team github organisation, welcome! :smile:

closed 2020-12-13 by renzenicolai

smunaut, 2020-12-13:
Tx :)