badgeteam/mch2022-badge-hardware #26: Schematic review

kliment, 2020-12-06:
Issues:

- [ ] ESP32_EN needs 100n cap to GND to avoid glitching
- [ ] R67/R68 should have higher value - at least 2k, to avoid large current source/sink if driven opposite ways by both signals
- [ ] PWR_LED should probably have a bulk capacitor
- [ ] Do the FPGA rails need power sequencing?

smunaut, 2020-12-06:
wrt to FPGA: Officially the 1v2 core of the FPGA must come before any of the other rails.
In practice, raising them at the same time never seemed to cause any issues.

kliment, 2020-12-06:
I don't see why we can't have the enable on the other vregs driven by the 1.2V regulator output. That gets us correct power sequencing for almost free.

zoe1337, 2020-12-06:
When we're adding caps to the ESP32 enable line, let's do the same for the STM32's nRESET lines!

smunaut, 2020-12-07:
wrt to R67/R68:

Using higher resistance reduces the maximum speed of the lines. Given how those leds are driven, you'd want your signal to have rising/falling edge well defined in time to preserve timing, so probably want a bandwidth ~ 10x the maximum rate, that would be 8 MHz here. With a 2k in series, that means the absolute maximum capacitance on the line would be 10pF. The U10 buffer has 4 pF input capacitance on its own, so any other parasitics need to stay low.

For the LED it might just stil work but for many other places that uses the same scheme, the frequencies involves are much higher and the bandwidth shouldn't be limited by a protection scheme.

The current is also not going to be _that_ high. Each driver has an output impedance, assuming they're are fairly strong it's be in the 25R range, plus the 2 47R, current is roughly 23 mA, the drivers should survive that without too much trouble. It can be bumped to 68R to bring it to 17.5 mA if more margin is wanted without much worries about bandwidth. But I guess 47R was chosen to reduce the BOM lines since it's used everywhere (including in places where the absolute values would matter)

renzenicolai, 2020-12-12:
Solved the first two issues.

3uF should be enough capacitance on the LED power line, I've even removed some caps.

FPGA is left in RESET state until it gets programmed, a long time after the power lines get stable. Power sequencing isn't going to be a problem in our case.

closed 2020-12-12 by renzenicolai