badgeteam/mch2022-badge-hardware #28: USB input protection

tachiniererin, 2020-12-06:
The USBLC6-4 (D10) ESD protection specifies a worst case clamping voltage of 10V between Vbus and GND.
The SY8079 (U1 and U2) DC/DC converters have an absolute maximum input voltage of 7V. The power switch SY6280 (U14, among others) only of 6V.

When hot-plugging a long USB-cable, parasitic inductance and bulk capacitance can easily cause transients that not just briefly violate the abs. max rating, but also have enough energy to cause permanent damage.

Either higher voltage rated parts would be needed, or additional input protection; either a back-to-back pMOS pair (at least 30-V rated), or an integrated solution like an eFuse or an OR-ing controller (the latter could also handle the battery power switching).

Also a resettable fuse in series with USB Vbus is a good idea to protect the upstream USB port; this board is meant for tinkering, and accidents do happen.

renzenicolai, 2020-12-12:
Could you look at LCSC for a cheap suitable fuse?

tachiniererin, 2020-12-14:
I haven't found anything which is readily available and cost effective unfortunately. But i'll try to sketch a discrete solution in the coming days which takes a bit more space but should cost less than an ic.

zoe1337, 2020-12-20:
you can add a capacitor to the pass pFET's gate-source, slowing it down and preventing inrush

renzenicolai, 2020-12-21:
Added a fuse on the Vusb input, together with a ferrite bead in chip size. Should cover this stuff together with the 10u+100n capacitors.

closed 2020-12-21 by renzenicolai