badgeteam/mch2022-badge-hardware #29: Flash voltage selection

tjclement, 2020-12-06:
I think the pull-up on ESP32's GPIO12 will cause the SPI flash to brown out, because the ESP32 will set it to 1.8v instead of the 3.3v that the WROVER B and E need; only the original WROVER works at 1.8v (and thus with GPIO12 pulled up)

https://www.espressif.com/sites/default/files/documentation/espressif_products_ordering_information_en.pdf
https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf ("strapping pins")
https://www.espressif.com/sites/default/files/documentation/esp32-wrover-b_datasheet_en.pdf
https://www.espressif.com/sites/default/files/documentation/esp32-wrover-e_esp32-wrover-ie_datasheet_en.pdf

sebastius, 2020-12-06:
Ah, same issue as we had on HackerHotel 2019? If i recall correctly you can burn a fuse to fix this, right?

martinling, 2020-12-11:
You can - the relevant fuses are `XPD_SDIO_REG`, `SDIO_TIEH` and `sdio_force`), in the "eFuse Controller" chapter in the ESP32 TRM: https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

But that's an extra step you'd have to do for every single unit, so it would be a lot better to find a way to avoid needing a pullup on that line.

renzenicolai, 2020-12-12:
Swapped ext i2c with i2s clk/data pins and made IO12 have a pull-down.

I do suggest we burn these fuses and the write-protect fuses on any other unwanted option, that way we can be sure no-one accidentally bricks their badge / stop malicious software.

closed 2020-12-12 by renzenicolai