badgeteam/mch2022-badge-hardware #30: FPGA: Try to respect diff pair for pmod

smunaut, 2020-12-06:
Diff pair on the FPGA are the same as IO site pair.

It wold be nice if the standard way (at least for ice40 fpga board and ulx3s and others ...) was met.
This means on the PMOD side:
 * 3 / 7
 * 2 / 6
 * 1 / 5
 * 0 / 4

And on the FPGA side it would be those pairs:
 * `IOB_2a` / `IOB_3b`
 * `IOB_4a` / `IOB_5b`
 * `IOB_8a` / `IOB_9b`
 * `IOB_0a` / `IOB_6a`  (this isn't a diff pair, or an IO pair but just the 2 other isolated IO site in that bank ...)

Preferably use `IOB_0a` / `IOB_6a`  for the 3/7 pair, mostly for compatibility with icebreaker-bitsy pmod adapter pinout and PMODs designed for it. Also if possible keep the `a` (which is positive diff pair) on the `0/1/2/3` side of the pmod.

renzenicolai, 2020-12-07:
Any chance you might be able to move the pins on the FPGA schematic around a bit, then do a PR?

smunaut, 2020-12-07:
Sure I can give it a shot. I won't get to it until the week end though.
Should I ping you here ?

renzenicolai, 2020-12-07:
yes please, I will try to fix most of the other issues before then

smunaut, 2020-12-12:
I was planning on working on this tomorrow morning if that's good for you ?

renzenicolai, 2020-12-12:
Yes, tomorrow morning works. Just now I've just pushed the fix for some of the reported issues, I will now leave the project alone until after you're done fixing the FPGA pinout.

Thank you very much for taking the time to improve this badge!

smunaut, 2020-12-13:
This is what I ended up with. 

[image]

Comments:
* I renamed the PMOD pins to follow the official naming (1 -> 4, 7 -> 10).
* `PMOD_10i` and `PMOD_4i` can be swapped for routing ease if needed
* The PMOD connector should be moved to the top rather than the bottom (and possibly to the side then if there isn't enough spcace on the bade top edge). A lot of PMOD with user interaction (button screens) need to have the "right side up" to work. I selected the pinout to try and make routing "easier".  Although it's still not great because Lattice pinout/padframe sucks and they didn't put pins in a single io site/diff pair next to each other :/

renzenicolai, 2020-12-13:
thank you :smile:

closed 2020-12-13 by renzenicolai