badgeteam/mch2022-badge-hardware #32: EMC considerations

martinling, 2020-12-11:
Hi, I'm that weirdo from the EMF badge team who cares about EMC.

So you may think this doesn't matter because it's just a camp badge, not a product you're trying to get tested & certified. But I think it's worth trying to get the basics right, and I'll explain why - along with some really simple & cheap things you can do to test & improve things.

Looking at the design, I can guess from experience what the main EMC issues will be. You have two switching regulators and they're going to shit noise everywhere. The root causes will be at the fundamental frequencies of the switching rate and the ringing of the switch waveform, and you'll see harmonics of both of those running up into the tens or hundreds of MHz.

Actually, if you look at the output of those regulators it will probably look pretty clean, because in a buck topology the inductor and output capacitor form an effective LC filter without any further help. But the big problem isn't on the output side; it's on the input side. The switching noise will flow back into the input supply unimpeded, and the PCB and the battery will then act as an antenna to radiate that noise.

Fortunately, the device is small enough to be a poor antenna at the frequencies that are likely to be involved, so it won't radiate particularly well. Until you plug in the USB cable, that is. Now you have a great antenna, that's a couple of metres long, and it's going to radiate common mode currents on it quite nicely in the VHF range.

Worse still, the lower frequencies are going to conduct back through the USB cable into the mains, and at camp, that's a bunch of wires strung around a field, which are going to make a decent HF antenna right down into the low MHz range.

So you've got a wideband noise source, which is going to be plugged in every evening to a very effective antenna system. And even if the noise power is small, if you multiply it by 3000 badges, that's equivalent to attaching it to an RF amplifier with a 35dB gain. It's going to make a big mess all over the radio spectrum, and although it'll might be hard to spot any particular problem as a result of that, it contributes to making a lot of things work poorly for everybody - particularly all the ham radio operators and other RF nerds at the camp.

So here's a list of what you can do to fix this, highest priority first:

- [ ] Add a common mode choke on the USB power lines. For the EMF badge we have part WCM3225F2SF-102T15.
- [ ] Add a common mode choke on the USB data lines. For the EMF badge we have part MCE1012E900FBP.

These will reduce common mode currents getting onto the USB cable, where they can radiate. They won't affect the normal flow of power and data since those are both balanced, differential mode currents. To add the choke to the power lines you will need to create a separate ground around the USB connector, and connect this to the system ground only through the choke. The chokes should be as close as possible to the connector, but not closer than the USBLC6-4 TVS array.

- [ ] Add footprints for RC snubbers around the switch nodes.

One of the root sources of the noise will be ringing of the switch waveform. You can kill this off at the source by adding a series R-C pair in parallel with each switch. On the SY8079 there are two switches - one between VIN and SW, and one between SW and GND. So add an R and C in series between VIN and SW, and also between SW and GND. The values can be tuned empirically by trial and error while looking at the switch waveform on a scope. Here's a good app note on the technique:
http://rohmfs.rohm.com/en/products/databook/applinote/ic/power/switching_regulator/buck_snubber_app-e.pdf

- [ ] Add an LC filter at the input of each switching regulator.

This will help attenuate the switch noise propagating back into the power supply. A shunt capacitor of a few uF, plus a series inductor of a couple of uH, should do the trick. At least put the footprints in - you can always fill the L with an 0 ohm resistor and leave out the cap, if you don't need them.

There are more things you can do to optimise things further if you want to, but the above points are what I'd expect to make the biggest differences for this design.

If you get the necessary footprints included then I'm happy to help with testing and tuning if you can ship me a prototype board. You don't need an EMC lab - everything needed can be done on the bench with a scope, a basic SDR if you don't have a spectrum analyser, and some homemade probes.

zoe1337, 2020-12-11:
I haven't tested this particular Silergy converter, but I was doing design validation of low power DC/DC converters and I don't think this 1-MHz converter is of particular concern.

> One of the root sources of the noise will be ringing of the switch waveform. 

This "ringing on switch node" is virtually nonexistent at these power levels. Since the FETs are small and are on the same die, the parasitics are negligible. See waveform cropped from datasheet:  
[image]

The switch node indeed has high dV/dt and it should be minimized, but with proper PCB layout it is a very small antenna. An RC snubber is absolutely unnecessary, and would only ruin efficiency.

You are correct that input current is discontinuous, and that's what the input capacitor is meant to help with. To suppress higher frequency harmonics, a 100-nF 0402 MLCC parallel to the 22-uF C4 cap is a good idea.

This noise is, however, differential mode, and a CM choke would do nothing to suppress it. Luckily, it only forms a loop antenna of a very small surface, just between the VBUS and GND lines. With shielded cables this is not a concern at all, and with some luck you might pass the legal EMI limits without further action. The input MLCCs are excellent at suppressing switching harmonics, especially if low ESL (small footprint) additional decoupling is added, as recommended in my previous paragraph.

That said, if so desired, another 22-uF MLCC can help, at little expense. Should a differential mode ferrite bead necessary, that needs to have rather high inductance and low enough DCR, so it doesn't come at a large tradeoff. Something like NFZ32BW210HN11L (22 Ohm at 1 MHz) or NFZ32BW6R8HN11L (6.8 Ohm at 1 MHz), from Murata, both available at LCSC. These eat into the efficiency of the converter thanks to the additional ohmic losses in the power path, also, these might require even more input capacitance, and could require turn-on voltage slope control to avoid potentially destructive excessive ringing from inrush current.

Adding a common mode choke on the USB data lines is a good idea for unrelated reasons.

martinling, 2020-12-11:
I'm not sure you can rely on that datasheet plot to say what the edges on the switch waveform will really look like. Often if you probe a switch node with a scope probe with the usual ground cable, it looks okay at first because the inductance of the ground wire is effectively filtering out the higher frequencies. But if you take off the ground cable and use a short wire wrapped around the probe shield (as in Figure 6 in the Rohm app note above), the bandwidth of the probe setup increases and suddenly the picture can actually look very different, especially close up in the corner of the waveform.

You may be right that the small FETs make the issue negligible - and if there indeed isn't any ringing on the switch waveform then of course you don't need the snubbing. But I would certainly test it to be sure, and my suggestion is to include the RC footprints at the prototype stage, so that it's possible to snub a bit if needed; you can always leave them unpopulated, and remove at the next board iteration if not used. A little bit of snubbing doesn't have to ruin efficiency - it will always decrease it, but adjusting the RC values allows for a continuous tradeoff between efficiency and noise, and often it only costs a negligible % or two in efficiency to clean things up a lot.

I agree that a low-ESL 100nF in parallel with the 22uF input capacitor will help with the higher harmonics of the switching frequency that arise from the high dV/dt of the switch. My experience though is that some of the harmonics will still show up in common mode currents on the cable, presumably just because all impedances are finite and loop areas non-zero. A DIY RF current probe, of a couple dozen turns wrapped through a split ferrite with the USB cable running through it, can show these up on an SDR or spectrum analyzer in the HF to VHF spectrum.

I put the common mode chokes first on the list, because having them on both pairs at the USB port will severely attenuate the main routes for **any** noise getting off the board and radiating from the cable, not just that from the switchers (although I expect those to be the main source of trouble in this design).

The use of a shielded cable is of limited help here because there isn't a conductive housing to connect it to at the device end which is separate from the system's power ground. At RF it's easy for it to end up just being an additional conductor for trouble.

renzenicolai, 2020-12-12:
Thank you for the suggestions!

Would you, by any chance, have Kicad symbols / footprints for the WCM3225F2SF-102T15 and MCE1012E900FBP parts?

martinling, 2020-12-13:
Sure, see #33.

For WCM3225F2SF-102T15 you can use the `L_Core_Ferrite_Coupled` symbol from the Kicad standard library. For MCE1012E900FBP, use the `L_Core_Ferrite_Coupled_Small_GND` symbol included in #33 with the footprints.

renzenicolai, 2020-12-18:
I've added the chokes around the usb connector. How would you recommend me to connect the shielding of the usb connector?

martinling, 2020-12-20:
> How would you recommend me to connect the shielding of the usb connector?

That's a great question for which I really just don't have a firm answer, I'm afraid - there are various flame wars you can read on the topic if you want to make your head hurt.

The basic issue as I see it is that for a board without a metal housing there's no great solution for what to do with the shield at the device end. It's a matter of balancing different considerations with opposing needs - for transients and noise immunity you tend to want minimum impedance from the shield to ground, because it's exposed to the environment and you want to short away everything that might couple onto it as firmly as possible. But that conflicts with minimising emissions - you want to avoid noise from your system ground getting out onto the shield, so you want some impedance in the way of it getting there.

Your basic options are:

1. Don't connect the shield at all.
2. Connect the shield directly to ground.
3. Connect the shield to ground through a capacitor and resistor in parallel.
4. Connect the shield to ground through a ferrite bead.

Each optimises for different things. Usually I hedge my bets and put a couple of footprints to GND, so that if the board fails EMC testing I have some options to try.

renzenicolai, 2020-12-21:
I'm going to connect both usb shield and usb gnd directly to my device ground, which will be a big copper fill on one of the layers.
In turn the USB power line will have a ferrite core in it's path, to at least limit the noise.

renzenicolai, 2020-12-21:
I've added an extra resistor in series between vin and the each of the regulators, together with a second capacitor at the input of the regulators. This should allow for implementing a low pass filter if needed.

martinling, 2020-12-22:
The ferrite & extra footprints for filtering at the regulators seem like a reasonable set of precautions against differential mode noise getting onto the USB power lines, which from an emissions perspective basically addresses conducted emissions at the lower frequency (< 30MHz) end of things.

I would still be worried about higher frequency common mode noise on the USB power lines, leading to radiated emissions directly from the cable in VHF & above, which the ferrite won't really help you with.

Common mode noise will be generated from ground/power bounce, on both the switching regulators and from switching currents in the FPGA & other digital parts, as well as poor return paths (i.e. large loop areas) for signals. So you'll see it at clock & switching frequencies and harmonics of them. You can minimise this by good layout and power distribution though, so if you're confident in getting that right, you may be okay to leave out the common mode choke. Personally I'd keep the footprint for it in as an option until able to test.

renzenicolai, 2021-01-11:
We've done what we can here I think. If you have suggestions for the current state of the project please open a new issue :smile:

closed 2021-01-11 by renzenicolai