badgeteam/mch2022-badge-hardware #36: Resistors as signal combiners have issues

Wilenzo, 2020-12-13:
In the schematic I see a few 47 Ohm signal combiner solutions, this is not really a good idea imho. For each time both inputs have a different state it draws about 33mA at 3.3V 

It's better to use diode logic for this: https://i.stack.imgur.com/ntAAj.gif

Whether you go for the AND or OR solution depends on the chips driving it: Do you normally have an active low or high state when it's doing nothing or do you have (internal) pull-ups/downs connected is mainly what makes you go for AND or OR.

The resistor must be chosen small enough so the slew-rate of the signal is sharp enough. It depends mostly on the input capacitance of the chip connected to the output of the diode gate combined with the diode capacitance and stray capacitance of the traces on the PCB. Put the logic close to the chip that's driven by the diode gate.

I've found the questionable resistor solutions on page 1 near the PCM5100, page 3, R19 & R20 (and R18 could even mean more than 66mA being wasted) and on page 4.

smunaut, 2020-12-18:
IMHO this is worse than resistors ...

Resistors only have high current when you screw up and drive from both side. It's a protection measure. Using diode logic means that 100% of the time you rely on a resistor for one of your edge and you're fighting that resistor when trying to drive the other state.

So you're increasing drastically power consumption during _normal_ operation while for the resistors, this only happens during contention / i.e. when you screw up and try to drive from both devices at once.

Wilenzo, 2020-12-18:
I agree it is not a perfect solution, a single logic gate would be much better. But using resistors this way is asking for problems.

The resistor solution is just not done because it could destroy chips that drive this contraption (excess pin current or even Vss/Vdd current if more pins are involved), it could destroy the chip that's being driven (signals midway between logic 0 and 1 can lead to destruction), and all that current draw can maybe lead to a bricked badge that is stuck in some brown-out loop or it could even lead to smoking components. And with resistors, a lot more EMI is generated because the trace to the chip that is in tri-state mode is driven hard by the other chip. With the diode solution this is much less of an issue.

Remember: User screw-ups are very common on prototyping boards and should be non destructive! The FPGA can only sink/source 8mA on its pins, with the resistors it is limited to 33/66mA... Almost 4 or 8 times too high! And if a user screws up, it might not become clear what the problem is because so many things are involved. Even with an oscilloscope it might not be very clear if you don't look closely at the voltage levels as the waveform stays intact.

The resistor in the diode configuration does not have to be a huge power consumer. It will probably be in the order of 5k for signals up to about 10MHz at 10pF load. (8pf for the input +2*1pF for the diodes, type: BAS16) But it's better to aim a bit lower, 3k3 should work nicely. With a 50% duty cycle signal this consumes 0.5mA at 3.3V. If the signals are 10% or 90% duty cycle it only consumes 0.1mA if the inactive state and the type of logic combination is chosen wisely. For high speed signals, lower the resistance or use logic gates.

smunaut, 2020-12-18:
(1) Not sure how you do your math but 5k with 10pF load has a 3 dB cut-off of 3.2 MHz. And you want your signal to remain "squarish" so at most I'd put 3dB f_c at the 5th harmonic. So for a 10 pF load, your resistor value is like 300 ohm to get a decent edge at 10 MHz. 

(2) EMI: The high current to fight that resistor are going to generate much more of a field than the small amount needed to charge/discharge the trace, even if the effective antenna length is doubled.

(3) The FPGA can drive/sink way more than 8 mA, that's just the max value for which its Vol/Voh and transitions times meet the specs.  I've caracterized the UP5k drivers ( part of the results : https://github.com/YosysHQ/icestorm/wiki/UP5k-Driver-Characteristics ) and among the test were driving the IO directly shorted to the rail for long period with no ill effects. Heck, we've even fed 3v3 to the 1v2 rail of that chip and it took it without dying ...

(4) Drivers are not perfect, the one from the UP5k are ~ 50ohm impedance, not sure about the other side, but even if they're damn strong they'd be like 25 ohm. In another issue it was recommended to bump 47R to 68R. So that'd be 3.3v into 211 ohm which is only 15 mA worst case for a persitent conflict. That's not going to kill any decent cmos driver.

(5) The diode logic also kills some use case. For instance, Imagine you want to use the PMOD port with the HDMI PMOD to transmit image and sound ... well now the fpga can no longer passively listen to the sound transmitted by the other chip to encode it on HDMI.  Or to generate an equalizer display from the music or whatever ...

Wilenzo, 2020-12-19:
(1) 300 Ohm is a good idea if you have asynchronous signals that need a certain steady state bit length or when you have some hard wired signals you need to be (almost) in sync with, but for clocked signals that accept a far from 50/50 duty cycle on the clock and were signals all go through the same type of diode logic, 3k3 is still fine in most cases. I've used something like this before without problems. With 5k on 10pF it takes 50ns to reach 63.2% of the voltage level, for most CMOS signals it works, but it's 70% you'll officially need to flip the state, so that's why I was aiming at 3k3 or lower for this. But signals will be spiky, true.

(2) True, didn't think of that. Although it might not be just as straightforward as the floating pin and trace add a lot of capacitance. The diode reverse capacitance is less than a pF while the floating trace and pin will double the drive current too due to added capacitance and the length/area of the "antenna" is doubled too. So it's not a simple manner of just current through the resistor.

(3) Ok, didn't know, but still, going beyond specs is a thing I would never recommend. 

(4) What about using just, let's say a 150Ohm, resistor on one of the driving chips and the other one hardwired? This way, one chip always has the overhand when things collide, so no mid VCC voltages on the lines. And it makes for a much better to trace fault without much side effects. It saves a resistor too and it does not make me want to vomit as much as the two resistor solution.

(5) True, might even be a useful case too!

renzenicolai, 2020-12-21:
Changed resistor combinations with one higher value resistor and a short, so that one device has priority.

closed 2020-12-21 by renzenicolai

smunaut, 2020-12-21:
:(

smunaut, 2020-12-21:
(1) Having inline resistor to take the edge off is always nice to avoid excessive ringing (also helps EMI).

(2) When the user does something wrong, it *SHOULD* fail, that's how the user will know something is wrong. Here, he might be stressing the drivers of one of the device not for minutes to hours, but potentially for days, weeks because he will never know something is wrong since "it works" ...

Wilenzo, 2020-12-22:
(1) Most of the other lines don't have resistors in series anyways, plus the impedance of the drivers themselves already takes care of ringing.

(2) It is NEVER a good idea to have logic fail on half the logic voltage level. This could DESTROY input circuitry of components. That's why you never see this construction in final designs, unless maybe when all of the inputs have schmitt-trigger functionality. But I do agree there is in half of the cases no feedback to the user now, that's a pity, but a mention in the docs and example code should be more than enough of a warning.