badgeteam/mch2022-badge-hardware #38: PWR_LED also on U10?

Wilenzo, 2020-12-13:
The LEDs can be damaged if U5 is in over-current mode and the LED data pin is being driven high. Power to U10 should also be connected to PWR_LED. The input of U10 should have a diode AND gate with a pull-up to PWR_LED so U10 is protected as well.

renzenicolai, 2020-12-18:
Change made: U10 is now powered by PWR_LED.

The datasheet for U10 (74LVC1G125) seems to suggest that it can't be damaged by an input voltage that's higher than the supply voltage, could you also have a look at (table 6 of) the datasheet? I'd have expected Vi to be expressed in "0 to vcc+0.5v" terms or the like if there was a risk of damaging the IC, but it states "0 to 5.5v", so regardless of Vcc it can handle 5.5v on the Vi pin.

Wilenzo, 2020-12-18:
Ah, this chip has no internal diode from input to VCC it seems, this allows for 5 to 3V level conversion. So it probably will be ok.

closed 2020-12-18 by renzenicolai