badgeteam/mch2022-badge-hardware #47: PMOD on top (or pinout such that pmod plug "right side up")

smunaut, 2021-01-11:
It's a comment I made on another issue, but not sure if this was considered.

Quite a few PMODs have "user interaction" elements (LED / button / screen / ...), and having the PMOD such that it is now, those would face toward the back ...

I would recommend either put the connector on the top of the badge, or, if for aestetic reason the actual connector has to be on the back, adapt the pinout so that PMODs are plugged with the right side up.

( Technically by putting 'GND' and '3v3' columns on each side you can even make it so that pmods can be plugged both way, assuming the user aligns it properly ... since it's a fpga, remapping pins is not an issue.  Not sure I'd recommend it though since it might be confusing )

sebastius, 2021-01-11:
The trouble we'll have anyway is that the PMOD connectors will be soldered by the visitors, so we need to make sure to have clear instructions on the board on what is what and how to mount it.

renzenicolai, 2021-01-12:
[image]

Viewed from the bottom of the PCB, connector is placed on the bottom side.

Is this good enough?

smunaut, 2021-01-12:
Electrically yeah, that looks good to me.

The silk doesn't follow the standard PMOD notation however. In the digilent pmod specs and usual labels, the pins are 1,2,3,4,7,8,9,10  and not 0-7.

And I would also add a note like "PMOD top side toward screen" or something like that.

renzenicolai, 2021-01-14:
Numbering has been changed.

closed 2021-01-14 by renzenicolai