badgeteam/mch2022-badge-hardware #49: 3.3V switching converter layout

martinling, 2021-01-14:
A few thoughts on this section:

 1. All the supply current for the 3.3V converter has to pass through this one little 0.2mm trace - make it thicker!

[image]

 2. I'd suggest moving C35 so that it's closer to U11 than C38. The idea (from #32) was to be able to put a smaller value capacitor there which would provide lower impedance at higher frequencies than the bulk capacitor. For that plan to work well, the smaller cap should be closer to the load, with the smallest possible loop area.

 3. I'd suggest rotating L3 90 degrees to put its pin 2 closer to U11 pin 3, to minimise the area of the SW33 net as much as possible. The smaller you can make it the less it can radiate.

 4. Personally I would thicken up the grounding around this section a lot - use  a polygon fill around the ground pins of the bulk capacitors, add more vias from there to the groundplane beneath, and add turn the ground ring around the whole section into a via fence. Similarly, you can put a polygon around the 3.3V pins of the inductor, the output caps, and the vias to the 3.3V plane.

kliment, 2021-01-14:
I'm on this

kliment, 2021-01-14:
Implemented mostly as suggested for both converters.

commit 1032ce3b (2021-01-14): Implement martling's suggestions re: converter layout. Fixes #49
closed 2021-01-14 by kliment