badgeteam/mch2022-badge-hardware #50: Long VIN trace with poor return path - likely EMI issue

martinling, 2021-01-14:
Most of the layout looks reasonable from an EMC point of view - the high speed stuff is mostly on the back side with a continuous ground plane directly under it.

What struck me as a problem is this long VIN trace, which I think is at risk of causing some radiated emissions:

[image]

It will have high frequency noise on it, potentially at quite high currents, because it's providing the input supply to both switching converters.

It's long enough to make a decent antenna.

And it doesn't have a good return path - VIN is referenced to GND, but the plane directly under the trace is at 3.3V. The GND plane further below it is a lot further away (assuming typical 4-layer stackup) and high frequency currents will try to stick close to the trac. They'll end up the using 3.3V plane beneath as the return path to the extent they can, but there's two slots in that plane near U5. There's also another gap in the return path where it cuts a corner through the antenna keepout zone in the top left of the board.

Can you avoid needing such a long trace? Or arrange to have it run over a continuous ground plane?

kliment, 2021-01-14:
We can't avoid it because we need to supply power to the LEDs in the top corner and the source of the power is either battery or USB, which are on bottom edge. I might be able to manage continuous ground but then it would have to run along the other side and be even longer. The 1V2 supply is placed to minimize vin trace length, whereas the 3V3 supply is placed exactly between its two highest-current users. I'll see what I can do. Do you have any recommendations?

kliment, 2021-01-14:
Solution: make long trace longer, give it continuous return path immediately below.

[image]

martinling, 2021-01-14:
That looks better to me, although I'm a bit wary of having a noisy high current trace running along a board edge as you have at the top there - I try to keep some ground all the way around the outside of a board and tie the ground fills with vias there too.

What I was just writing before you posted that, is that if you can't change the basic arrangement then I suggest to add some small capacitors (say 10nF) between 3.3V and GND, with adjacent vias to the 3.3V/GND planes, at the following places:
  1. At the ends of the trace.
  2. At points in the middle where loads are connected.
  3. On either side of any breaks in the 3.3V plane under the trace.

That would allow high frequency currents to use the 3.3V plane as a a return path.

My other thought was that LC filters at the switcher inputs would help keep noise off the trace in the first place.

kliment, 2021-01-14:
I added ground all the way around the trace along the edges. There's also a groundplane on the layer immediately below everywhere except a tiny bit at the top left corner.

renzenicolai, 2021-01-14:
Hello y'all, I'd like to see a solution that doesn't affect the looks of the badge as much. This new trace you made is clearly visible to the user unfortunately. Is there a way to hide it a bit more?

martinling, 2021-01-14:
@renzenicolai, was it OK visually where it was before? You could leave it there (but clean up the corner in the top left) and add the capacitors discussed above to improve the high frequency return path. Using the 3.3V plane as a return path will add some noise onto the 3.3V supply, but I don't think you have any particularly critical analog stuff on that rail - the IMU sensors and the audio codec do some internal regulation IIRC.

@kliment I wouldn't worry too much about minimising the length specifically - what matters is the loop area. As long as the return path is close (and getting it on the adjacent layer makes it like 5x closer for a typical stackup) then the loop area is small even if the trace is long.

martinling, 2021-01-14:
Just seen the new version - looks good to me now.

commit 68cb4859 (2021-01-14): Reroute VIN and add a reference plane immediately below VIN trace everywhere. Fixes #50
closed 2021-01-14 by martinling