badgeteam/mch2022-badge-hardware #89: Wish: JTAG debug interface

stappersg, 2022-08-05:
Some IRC log, MCH2022 was already over:
```text
17:46 < zgreg> the workflow for getting python apps onto the badge and running
    and debugging them never really worked for me
17:48 < zgreg> but you can't debug native esp32 applications without additional
    hardware, or can you?
17:49 < JulianScheffers> The JTAG is inaccessible unfortunately.
17:49 < JulianScheffers> I usually have to do print debugging.
17:49 < zgreg> that's something that always seemed odd to me about the
    esp8266/esp32 ecosystem. proper interactive debugging was mostly seen as
    something esoteric that isn't really needed...
17:50 < TomClement> zgreg: This is something that became accepted in the Arduino
    era, imo. Luckily for ESP32 there’s a nice JTAG interface you can use,
    although not all projects expose it
```

Now my wish:   The `rp2040` is also a _JTAG probe_.

That wish smaller:   The PCB for a next  _esp32-rp2040-combo_ has wires, copper tracks, that make it possible that the `rp2040` can be used as _JTAG probe_ to connect the `esp32` JTAG interface.

smunaut, 2022-08-05:
The problem is not that the pins/pads are not accessible ... is that every single IO on the ESP32 is used for something, so the JTAG has to be disabled to make room for them.

stappersg, 2022-08-05:
If a JTAG debug interface for current PCB is possible, please say so.  After all for (re)flashing the board is the USB-connector already being used (`rp2040` "talking" to `esp32`).

smunaut, 2022-08-05:
It's not possible ... that's what I said above. And no PCB revision will change that, there is just not enough IO pins on the ESP32 module to have all the functions of the badge _and_ JTAG.

The reflashing doesn't use JTAG at all, it uses the ESP32 rom bootloader and goes through the serial connection.

stappersg, 2022-08-05:
Oops

[image]

> The reflashing doesn't use JTAG at all, it uses the ESP32 rom bootloader and goes through the serial connection.

Acknowledge on that.

And I admit that I don't know if the _SPI connection_ between the _esp32_ and the _rp2040_ is  20% or 80% of a _JTAG connection_.

smunaut, 2022-08-05:
Huh ?

(1) There is no SPI connections between the ESP32 and RP2040.
(2) I have no clue what you mean by "being 20% or 80% of a JTAG connection" 

The only thing the RP2040 can do is basically reset the ESP32 and send the special char through the serial (as in: UART/RS232) link to go to bootloader mode and flash the thing. But obviously in that mode, the bootloader is running, no user code, so you can't use that to debug your code or anything like that.

stappersg, 2022-08-05:
> There is no SPI connections between the ESP32 and RP2040.

I stand corrected, not SPI but I2C. Source, the blockdiagram from https://badge.team/docs/badges/mch2022/hardware/

[image]

New version of the wish:
In a next  _esp32-rp2040-combo_ is it possible that the `rp2040` can be used as _JTAG probe_ to connect the `esp32` JTAG interface.

a2800276, 2022-08-05:
> it's that every single IO on the ESP32 is used for something

Tragically, that's not the case, there are ~9 unconnected IOs. We really should at least route everything to board edge for future badges...

But the JTAG pins _are_ occupied. 

@stappersg you could theoretically cut some traces and use JTAG but that would kill FPGA, SD Card and I2S functionality (12,13,14 are necessary for JTAG, quick dirty google search, may not be correct.) 

JTAG is one of few peripheral that can't be mux'ed to other pins, so the rp2040 would also not be able to access those specific pins, i.e. there is no possible firmware fix.

smunaut, 2022-08-05:
What are you talking about ?
What 9 IOs ?

- Pins 17-22 are NC on the ESP32-WROVER-E modules used on the badge (they're used internally for the flash inside the module)
- Pins 27/28 are just NC on that series on module

So I'm not sure where you see 9 unconnected IOs.

a2800276, 2022-08-05:
Wishful thinking :)    You're absolutely correct, I'm used to WROOM, which has a single nc pin. What were they thinking with that module?

smunaut, 2022-08-05:
AFAIU the modules that have those IOs don't have embedded flash ... and then you need to wire up the flash to them so it doesn't give you more IOs, it's just if you want your flash outside the module.

renzenicolai, 2023-04-22:
smunaut is right, ESP32 uses specific pins for JTAG which are NOT connected to the RP2040 on the MCH2022 badge. If you really want to use JTAG on your badge you can find the connection points here:

TDI is GPIO12
TCK is GPIO13
TMS is GPIO14
TDO is GPIO15

GPIO12 is used as I2S left-right clock
GPIO13 is used as I2S data
GPIO14 is used as SD clock
GPIO15 is used as SD CMD

So if your app does not use the audio output and the SD card you can safely ignore those peripherals and use JTAG. The I2S pins are available on testpads but the SD card pins are not so I recommend you connect small wires directly to the pads of the ESP32 module.

Good luck!

closed 2023-04-22 by renzenicolai